woohyeok-choi/ABC-Logger #12: App crash during using Facebook

suwon419, 2020-01-24:
**Describe the bug**
A clear and concise description of what the bug is.

App crash occurs and message pop-up.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

- Browse with Facebook app

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-01-25:
From v 0.9.3, ABC Logger includes Firebase' crashlytics. This may be helpful for a true reason of this bug.

woohyeok-choi, 2020-01-26:
I found several problems potentially related to this bug.
1) For collection of Bluetooth, some bluetooth device does not have any name, leading to null pointer exception.
2) For collection of call log and messages, there are a case that does not have a number (but I don't know why).

These two issues will be corrected in v 0.9.4.

closed 2020-01-29 by woohyeok-choi