woohyeok-choi/ABC-Logger #22: Delayed survey activity loading showes unintended text color.

suwon419, 2020-01-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

After I loaded the survey page through clicking the ESM noti and wait the next noti, the color of pre-loaded survey keep showing blue, which means that it can be answered.
I cannot save the answer when clicked the survey (it is normal and intended result)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

[image]

[image]

woohyeok-choi, 2020-01-30:
You mean, once open ESM response screen (or activity) by clicking a notification and wait for a while until timeout, then you cannot answer the opened ESM even if still it seems like available to answer (i.e., blue and bold), right?

If then, it is an intended behavior. It is difficult to update UI in real time (i.e., counting timeout and update UI...)

woohyeok-choi, 2020-01-30:
If I misunderstand this, please come to my office :)

closed 2020-01-31 by woohyeok-choi