woohyeok-choi/ABC-Logger #25: App dies when click "Config"

suwon419, 2020-02-08:
**Describe the bug**
A clear and concise description of what the bug is.

The app crash occurs when click Config button.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

Android 6.0.1 and GalaxyS5

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-02-09:
I found that this problem only occurs on Android 6.0.1. This may be due to incomplete compilation of Kotlin code, especially a static function (companion object in Kotlin).  

The putting annotation, @JvmStatic, in the static functions may be resolve this problem (but I'm not sure). BTW, users of Android 6.0.1 should be excluded till I definitely resolve the problem.

woohyeok-choi, 2020-02-10:
Alternatives for resolving this problem is to change Kotlin's object into a class, and use this class as a singleton class. I guess this option does not need additional test for reproduction :) but it requires some time for implementation.

closed 2020-10-20 by woohyeok-choi