woohyeok-choi/ABC-Logger #47: UI of the app has not been loaded for a long time.

suwon419, 2020-02-28:
**Describe the bug**
A clear and concise description of what the bug is.

After turn on the app, the app starts log-in the Google ID.
The main page of ABC Logger (Survey list) cannot be loaded at the time.
Also, the Config page cannot be loaded.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

This is not a quite clear path of reproducing.
Since many of LTE-sync users report this issue, I guess that the simultaneous sync and load can lead this problem.
We have already solved that the survey item disappearing issue even before timeout. That problem may cause by the sync. I think that the sync can interfere UI loading after solving the timeout issue.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

suwon419, 2020-02-28:
[image]

woohyeok-choi, 2020-07-10:
Releated to Isssue #48

soowon-kang, 2020-12-04:
fixed in 0.9.9-iota

closed 2020-12-04 by soowon-kang