woohyeok-choi/ABC-Logger #54: The timestamp of Key log table should be modified.

suwon419, 2020-03-25:
**Describe the bug**
A clear and concise description of what the bug is.

The current version's timestamp in Key log table gets AccessibilityService.eventTime.
-- https://stackoverflow.com/questions/55030433/android-accessibilityservice-geteventtime-time-format

This means that the timestamp cannot be easily replaced.

The following links are some helpful information to debug this issue.
-- https://developer.android.com/reference/android/view/accessibility/AccessibilityEvent#getEventTime()
-- https://developer.android.com/reference/android/os/SystemClock?hl=en
-- https://github.com/zelatore/KeyLogger/blob/update_keytracker/app/src/main/java/kr/ac/kaist/jypark/keylogger/KeyTrackingService.kt

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

I think that System.currentTimemillis() would be better for this,
but I'm afraid that the exact event time (key touch) can be slightly wrong.
I'm not sure that the exact event time and the currentTimeMillis always show the same result. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

cheonsol-lee, 2020-04-03:
It needs a timestamp to set standards like UnixTimeStamp.

suwon419, 2020-07-10:
Change the timestamp format as UNIX timestamp.

closed 2020-07-10 by suwon419