woohyeok-choi/ABC-Logger #56: The time at "isPosted=false" is wrong.

suwon419, 2020-04-09:
**Describe the bug**
A clear and concise description of what the bug is.

The notification collector should collect the time when each post turned on/off.
The "isPosted" value is related to the above statement.
In the "onNotificationRemoved ()" function, the "isPosted" value set as false.
However, "val postTime = sbn.postTime" is not working correctly at this time.
We intended the collector should log the time of the post off, but it logged the time of the post on again.
Thus, there exist all pairs of isPosted=true/false and the same timestamp (=the time of the post on).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

We should know the time of each post off.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

suwon419, 2020-07-10:
The original "isPosted=false" is fixed, but a related and unexpected issue occurs now.
We cannot match the notification pairs (isPosted=true and false), which means that which notification was removed from the notification bar even if we found one.
I think we need a UUID for each notification.

A notification may not be removed when the app showed it continuously.
To explain this, I assume that Slack received 3 msg from another colleague.
The notifications of slack may occur three times, but removing them occurred only once when the user clicked the msg.

woohyeok-choi, 2020-07-15:
Actually, this is not a problem. As  you know, each notification has its ID (like, NotificationManager.notify(int, Notification) ), and the previous notification goes away (but, actually not removed; just updated) if the app deliveres new notification with the same ID. Maybe we can add a field saying notification ID and TAG

woohyeok-choi, 2020-10-13:
Fixed

closed 2020-10-13 by woohyeok-choi