woohyeok-choi/ABC-Logger #60: (Data traffic) The collected data sometimes show the negative values, which is wrong.

suwon419, 2020-07-10:
**Describe the bug**
A clear and concise description of what the bug is.

In the database, we collected about a 2-month long dataset.
When I checked the data-traffic table, there are several negative values of sending/receiving bytes of network communication.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Cannot understand how the data bytes become negative for both sending/receiving cases.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

suwon419, 2020-10-28:
@woohyeok-choi 
I found some error records only in the alpha version data. (No weird record is founded in the beta~epsilon version.)
You can find them by using the below code.

db.datum.find({
    $and: [{
            datumType: { $exists: true }
        }, {
            subject: { $exists: true }
        }, {
            datumType : "DATA_TRAFFIC"
        }, {
            "value.fromTime": {$gt: 1600439766323}
        }, {
        $or: [{
            "value.rxBytes": {$lt: 0}
        }, {
            "value.txBytes": {$lt: 0}
        }, {
            "value.mobileRxBytes": {$lt: 0}
        }, {
            "value.mobileTxBytes": {$lt: 0}
        }]
        }
    ]
}).sort({
    datumType: -1,
    subject: -1,
    timestamp: -1
}).limit(10)

woohyeok-choi, 2020-10-28:
@suwon419 I found negative values are reported in a version of 0.9.9-delta.

commit d7db14de (2020-11-21): Try to fix #60

the data traffic collector tried to collect the corresponding data many times at the same time.
closed 2020-11-21 by soowon-kang