woohyeok-choi/ABC-Logger #69: (Notification) Tag must not be null

suwon419, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

Cannot collect any data.
Error msgs occur repeatedly.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the "Notification" collector (Status -> ON)
2. Error occurs in one minute. (See screenshot)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Smartphone (please complete the following information):**
 - Device: Note9, SM-N960N
 - OS: ANDROID10
 - App ver: 0.9.9-alpha

woohyeok-choi, 2020-10-20:
It will be fixed at a commit, #54e45643124a43e83e0ca8dd8d7cfdd2b3c23422

suwon419, 2020-10-21:
problem solved.

closed 2020-10-21 by suwon419