woohyeok-choi/ABC-Logger #70: (CallLog / Message) Permission Denial

suwon419, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

Cannot collect any log

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the collector (Status ON)
2. Error occurs in any minute (less than 5 min)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android 10
 - App ver: 0.9.9-alpha

**Additional context**
Add any other context about the problem here.

suwon419, 2020-10-20:
The errors repeatedly (about 15-min period) occur even if turned off the Call Log and Message collectors.

woohyeok-choi, 2020-10-20:
It is may be due to missing permissions.

suwon419, 2020-10-21:
problem solved.

closed 2020-10-21 by suwon419