woohyeok-choi/ABC-Logger #71: (Collector) instant message occurs when click "Availability" in Collectors

suwon419, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

If click "Availability" of a collector that has already set as "Available",
an instant message occurs. (see screenshot)

**To Reproduce**
Steps to reproduce the behavior:
1. Click the "Availability" button in a collector (e.g., Call Log)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android10
 - Version 0.9.9-alpha

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-10-20:
Does this problem occurs in any collector?

suwon419, 2020-10-20:
It seems the message occurs only in a few collectors that do not need to change any "availability"

woohyeok-choi, 2020-10-20:
It will be fixed in the next release (0.9.9-gamma)

suwon419, 2020-10-28:
confirmed in epsilon version

closed 2020-10-28 by suwon419