woohyeok-choi/ABC-Logger #72: (Embedded Sensor) ID is higher or equal to internal ID sequence

suwon419, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

During data collection,
error message occurs when removing the collected records by clicking the "Data size on Disk" button.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the Embedded sensor collector
2. Click "Data size on Disk" to remove the collected records
3. Error message occurs in a minute.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android10
 - Version 0.9.9-alpha

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-10-21:
@suwon419 I cannot reproduce this issue again. Does this problem occur again in recent days?

suwon419, 2020-10-22:
[image]

yes.

embeded sensor 켠 상태로 데이터 좀 모으던 중에 바로 데이터만 지우면 에러메시지 뜹니다.

suwon419, 2020-10-23:
still occurs in delta version
[image]

woohyeok-choi, 2020-10-26:
The reason why this error log was not uploaded in the Firebase Crashlytics seems to be due to intergration of Firebase with Slack. 
Now, I disconnect Firebase with Slack; please reproduce this error.

suwon419, 2020-10-26:
@woohyeok-choi I did. please check again.

woohyeok-choi, 2020-10-26:
@suwon419 Thanks to reproduction; I change some codes in a 0.9.9-epsilon for that. Please check this error again in that version.

woohyeok-choi, 2020-10-26:
Shit...Again!!!

woohyeok-choi, 2020-10-26:
I found a possible reason for that issue; differetn threads are launched on writing data and deleting database files. 
To resolve issues, I have two options (and corresponding problems)

1.  Make all operations thread-safe with mutex
This option seems promising, but it may result in bottle-neck for massive data writing operations, if we collect massive sensor data (like Polar H10).

2. Make deleting DB files unavailable when any collector is on (therefore, possibly data is writing) or sync.
This option is transparent to users,  but yeah, somewhat burdensome. 

I think a second option is easier to be implemented and less-vulnerable to possible errors.
So, in the next version, I disallow to delete DB files while any collector is operating or synchronization is in progress.