woohyeok-choi/ABC-Logger #73: (Config) Data size on Disk showed very higher value than the actual size of the app.

suwon419, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.

It seems that the value of Data size on Disk increases too fast (about 100MB/min).

I turned on every collector (except for polar, survey, call log, message, notification, wifi, location) that works well for now,
and the number of records from each collector increases normally.

I checked the actual size of the app in "저장공간" of ABC Logger,
and it showed a very small size (데이터: 2MB) for now (about 20 minutes collection).

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android10
 - App version: 0.9.9-alpha

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-10-20:
Let me know whether this is due to just representation error or not.

suwon419, 2020-10-21:
In-app data size value is a bit different from the android system's value (37MB vs 38MB).
But, I think it seems good enough.

closed 2020-10-21 by suwon419