woohyeok-choi/ABC-Logger #77: (Media) no data collected.

suwon419, 2020-10-26:
**Describe the bug**
A clear and concise description of what the bug is.

There is no data in DB.

**To Reproduce**
Steps to reproduce the behavior:
1. Capture screenshot, Use camera apps
2. But no data collected.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android10
 - Version delta

**Additional context**
Add any other context about the problem here.

woohyeok-choi, 2020-10-26:
This is because the time column of images or videos is formatted as a seconds, not milliseconds. It will be fixed at the next version

suwon419, 2020-10-28:
confirmed in epsilon version.

closed 2020-10-28 by suwon419