woohyeok-choi/ABC-Logger #79: (Message/DB) The timestamp in DB is weird.

suwon419, 2020-10-28:
**Describe the bug**
A clear and concise description of what the bug is.

Message's timestamp in DB represents the Year 2038.
Maybe, there are some misunderstandings when checking the #77 issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

db.datum.find({
    $and: [{
            datumType: { $exists: true }
        }, {
            subject: { $exists: true }
        }, {
            datumType : "MESSAGE"
        }, {
            "subject.groupName": "099e-tndnjs"
        },
        {
            "timestamp": {$gt: 1600439766323}
        }, 
    ]
}).sort({
    datumType: -1,
    subject: -1,
    timestamp: -1
}).limit(10)

woohyeok-choi, 2020-10-28:
Can you check this issue occurs in CallLog or Media? Because Message, CallLog, and Media collectors have same collection mechanics; so, if there is any problem amount those collectors, rest of them may have same problems.

suwon419, 2020-10-28:
Media works well but the same issue is founded in CallLog.

soowon-kang, 2020-11-11:
will be fixed in zeta

closed 2020-11-11 by soowon-kang