woohyeok-choi/ABC-Logger #88: (BoxStore) use closeThreadResources() to avoid finalizing recycled transactions

soowon-kang, 2020-11-22:
**Describe the bug**
A clear and concise description of what the bug is.

W/Box: Skipping low-level close for read-only cursor (non-creator thread '')
W/Box: Destroyed recycled transaction from non-owner thread ''

**Smartphone (please complete the following information):**
 - Device: Note9
 - OS: Android9
 - Version 0.9.9-eta

**Additional context**
Add any other context about the problem here.

soowon-kang, 2020-11-22:
The below link may be a good starting point.
https://github.com/objectbox/objectbox-java/issues/196

LeoK987, 2021-02-01:
I also noticed this warning recently.  on Android 9.  I went in to every thread and did closeThreadResources(), so it seems to have got rid of the warning.